utat-ss/HERON-lib-common #15: Clean up the branches on lib-common

vogeldylan, 2018-03-28:
Not sure which features are still under development; Cleaning up the branches would help to clarify this.

siddMahen, 2018-03-30:
Cleaned up all the branches I've pushed. Only remaining branches are from PAY @brunofalmeida but I think they're still being used.

brunofalmeida, 2018-03-30:
I'm currently working on `pay-can-dev-2`, will create a PR after testing it with what I'm working on in the `pay` repository.

brunofalmeida, 2018-03-30:
@siddMahen Have you considered deleting `code-cleaning` and `dev_branch_demo`?

siddMahen, 2018-03-30:
Git tells me that the branch contains commits that aren't referenced in any other branch, so I want to review those before deleting

siddMahen, 2018-03-30:
Same with `uart-wip`.

brunofalmeida, 2018-04-01:
What are the `obc-dev` and `uart-wip` branches intended for? They haven't been committed to since September.

siddMahen, 2018-04-01:
@brunofalmeida I think `obc-dev` is Ali's. Git tells me I shouldn't delete `uart-wip` because it contains commits that aren't referenced in any other branch. I'm pretty sure it can be deleted, but I want to make sure I'm not losing anything important. Will figure this out in a little bit.

siddMahen, 2018-04-01:
I think for the time being, the branch list is sufficiently small that it won't matter anyways.

siddMahen, 2018-04-01:
@vogeldylan Can I close this? I think we're basically done.

vogeldylan, 2018-04-02:
Done.

closed 2018-04-02 by vogeldylan