utat-ss/HERON-lib-common #18: Transition to assertion-based test code

siddMahen, 2018-04-01:
The current set of examples is still a useful way to test functionality, but we should begin moving towards a more standardized testing framework based on assertions. The examples we write tend not to test edge cases or rare behaviour because they represent typical use.

I'm thinking of something like:

```
lib-common
 └── network
 └── network
     └── client
```

```
#include <assert.h>

int main() {
    
}

brunofalmeida, 2018-04-01:
Would `assert.h` and `assert.c` be custom-made assertion code or a standard library? If it's a standard library, how would we see the results since it uses `stdout`?

brunofalmeida, 2018-04-01:
Would the `main()` function only be in `run_tests.c` if you just want to run that and have it execute all the tests? If you put a `main()` in each file, wouldn't you have to run each individually?

siddMahen, 2018-04-01:
@brunofalmeida It would have to be custom for exactly the reason you brought up; `stdout` just doesn't work like it regularly does, or at least I don't know how to make it compatible with UART printing.

siddMahen, 2018-04-01:
@brunofalmeida I think the `run_tests.c` code would run locally, on your machine. It would upload each program to the AVR, listen to the UART output, load the next test, rinse and repeat. It would also measure program size and speed.

vogeldylan, 2018-04-02:
What's the first example/test case we can develop for this? Might as well start some initial tests.

siddMahen, 2018-04-02:
@vogeldylan A great first test would be to make sure that `init_uart` and `init_can` and the like correctly set all the register values of the 32m1.

siddMahen, 2018-04-02:
If many people contributed to these tests, it would also make sure everyone knew how UART and CAN worked at the internal level, which might be good in general. This would require reading the current UART/CAN code, reading the documentation, and making sure everything is kosher.

bonnieweng1T9, 2018-04-03:
@jackberezny and I can take on this as well!

bonnieweng1T9, 2018-05-06:
@BrytniR  @JasonRocck  @jackberezny @mercury0524  This is the assert based testing that we discussed today with Sidd and Bruno, and one of the major projects we will be looking into after heartbeat.

siddMahen, 2018-05-15:
In the future, the harness could also be modified to report program size info, memory usage info, etc

brunofalmeida, 2018-05-16:
How is it possible for the test harness to read the 32M1's memory contents? The harness can also read UART output right?

siddMahen, 2018-06-05:
See #28 for on-going work towards this goal.

siddMahen, 2018-06-08:
Recently merged #28, which addresses almost all of our testing needs. @vogeldylan @alihaydaroglu

The only remaining aspect of testing which isn't incorporated into libtest is EEPROM testing. This can be added at a later date. I wanted to commit libtest to master quickly so we can begin writing tests to prep for qual; also I think it's more of a nice-to-have feature.

closed 2018-06-08 by siddMahen