utat-ss/HERON-lib-common #34: Ports Configuration

Tahreeem, 2018-06-15:
Line 29 of harness.py:
```
tail = tail[:-1] + str(int(tail[-1]) + 2)
```

That + 2 isn't accurate for my computer and might not be for other ones too. So COM8 and COM7 were the ports I had. Inputting COM8 (the programming port for mine) gave a result of COM10 from that line which was inaccurate for me. 

That line would need to be modified for each person's computer?

siddMahen, 2018-06-15:
@Tahreeem is the UART port always the programming port plus one? For example, if you're programming on COM7, is the UART port always COM8? If so, I can check to see whether the code is being run on Windows or on macOS, and do the right thing.

siddMahen, 2018-06-16:
@alihaydaroglu tells me that often the UART port is one less than the programming port. I'm going to add an optional `-u` argument which allows you to specify a UART port; if you don't supply `-u` it'll check whether you're on Win or macOS and make an informed guess.

Tahreeem, 2018-06-16:
@siddMahen The UART port is one less than the programming port for mine. But I'm not sure if all Windows computers are formatted that way. So for now, it might be better to have the user specify the UART port like you're suggesting.

siddMahen, 2018-06-16:
@Tahreeem I'm out of the country right now; could you test the changes I'm about to make on a board and let me know if there are bugs?

Tahreeem, 2018-06-16:
@siddMahen Sure thing! I can do that tomorrow since I'd be downtown.

siddMahen, 2018-06-16:
Fixed in #36

closed 2018-06-20 by siddMahen