utat-ss/HERON-lib-common #37: TypeError: unicode strings are not supported, please encode to bytes

Tahreeem, 2018-06-16:
Line 59, 66, 70, 225 produce that error on Windows.

The fix is simple. For example for Line 59:
``ser.write("COUNT\r\n")``
Changing it to the following fixed the problem:
``temp = "COUNT\r\n"``
``ser.write(temp.encode())``

siddMahen, 2018-06-17:
@Tahreeem what happens when you change `ser.write("COUNT\r\n")` to `ser.write(b"COUNT\r\n")`? I believe this should fix the problem.

siddMahen, 2018-06-17:
Also, I doubt its a Windows error; most likely a python 3 error.

siddMahen, 2018-06-20:
@Tahreeem what's the status on this issue? Did the above comment work?

Tahreeem, 2018-06-20:
@siddMahen I downloaded Python 2 today and didn't get that error so I'm guessing that was a Python 3 related problem.

siddMahen, 2018-06-20:
It's definitely a python 3 related issue; could you try running the code with Python 3 and the changes I suggested and let me know if it works?

Tahreeem, 2018-06-20:
I'll be in the lab on Thursday again. Will try with Python 3 too then and let you know of the errors tied to that.

siddMahen, 2018-06-21:
@Tahreeem Let me know if this works

Tahreeem, 2018-06-24:
I tried ``ser.write(b"COUNT\r\n")`` format yesterday but still got the same error: ``ValueError: invalid literal for int() with base 10''. I left windows at that...

siddMahen, 2018-06-24:
Ok, but you didn't get the error `TypeError: unicode strings are not supported`, so that's an improvement. The invalid literal business is a windows problem, unfortunately.

Tahreeem, 2018-06-24:
Hmm yes, the TypeError is resolved at this point.

siddMahen, 2018-06-24:
To confirm, the TypeError is gone when tested with Python 3?

Tahreeem, 2018-06-24:
Yes. Both formats resolve the TypeError.

siddMahen, 2018-07-20:
Fixed in #43 .

closed 2018-07-20 by siddMahen

palaniappanece, 2019-07-24:
import time
import serial

print("Starting program")

ser = serial.Serial("/dev/ttyUSB0", baudrate=9600,
                    parity=serial.PARITY_NONE,
                    stopbits=serial.STOPBITS_ONE,
                    bytesize=serial.EIGHTBITS
                    )
time.sleep(1)
try:
    temp="hello world"
    ser.write(temp.encode())
    #ser.write("Serial Communication Using Raspberry Pi\r\n")
    #ser.write("By: Embedded Laboratory\r\n")
    print("Data Echo Mode Enabled")
    while True:
        if ser.inWaiting() > 0:
            data = ser.read()
            print(str(data))

except KeyboardInterrupt:
    print("Exiting Program")

palaniappanece, 2019-07-24:
this program works in python3 to read and write the data from the serial port

palaniappanece, 2019-07-24:
library is sudo apt-get install python3-serial