utat-ss/HERON-lib-common #39: Running test harness

RebeccaShemesh, 2018-06-18:
This is how I found the ports I needed to use:
[image]

This is how I ran test harness:
[image]

And this is the error we kept getting. We tried playing around with it but aren't too sure what it means:
[image]

siddMahen, 2018-06-19:
Hmm that's very strange. The way the test harness works is that the 32m1 tells the harness how many tests it going to run, this is called the `count`, and then the harness tells it exactly when to start each test, and it does this `count` number of times. It seems like the test harness is having trouble parsing the `count` value returned by the 32m1.

This seems like it might be a problem with the UART on the board; were you using new boards?

Also, was this run on a Windows or a Mac? @RebeccaShemesh

RebeccaShemesh, 2018-06-19:
We managed to get arithmetic and print to run, but when we tried spi this came up:
[image]

RebeccaShemesh, 2018-06-20:
Also, we realized that the previous issue we had was because none of the main files (also print and arithmetic) had init_uart() in main.

siddMahen, 2018-06-20:
init_uart is already called by the testing library internally; it shouldn't be required to print anything

siddMahen, 2018-07-20:
@RebeccaShemesh closing this now because I think it's been resolved. Let me know if it needs to be opened again for whatever reason.

closed 2018-07-20 by siddMahen