utat-ss/HERON-lib-common #48: Watchdog timer and heartbeat integration

brunofalmeida, 2018-07-17:
On the `watchdog` branch:
- The `watchdog` library needs to be cleaned up and documented more
- The `func_tests/watchdog_test` test needs to be cleaned up, run on its own, and verified for correctness
- You need to develop a new test that integrates watchdog with heartbeat, i.e. use watchdog to reset the microcontroller and heartbeat to recover the state

**TODO** - need to add Andy to GitHub org and assignees

brunofalmeida, 2018-08-18:
It's the watchdog development task

jackberezny, 2018-10-01:
Points one and two should be be resolved when https://github.com/HeronMkII/lib-common/pull/76 is merged. Note that the func_test is now a part of the test harness suite.

brunofalmeida, 2018-11-21:
The core watchdog library is done, see #69 for heartbeat integration.

closed 2018-11-21 by brunofalmeida