utat-ss/HERON-lib-common #57: RFC: Qual development integration tracking

siddMahen, 2018-07-25:
I want to know what work was done in other branches leading up to, and perhaps slightly after, qual. Of this work, what's been merged into master, and what's left to integrate?

\cc @dhanyaa97 @brunofalmeida @bonnieWeng @Tahreeem @jackberezny @RebeccaShemesh

jackberezny, 2018-07-26:
For assert tests, this is what I know:
- CAN assert is completed and merged to master (work done on assert_can branch)
- SPI assert (I believe) has also been merged to master (spi_tests branch)
- UART assert was successfully tested on qual_assert before qual, but looks like it is still a WIP on the uart_tests branch now
- The qual_assert branch was to compile all the assert tests on one branch for qual, but I don't think that branch has a use anymore

brunofalmeida, 2018-07-26:
Yeah, from what I remember we demoed `qual_assert` at qual but I agree with Jack that it doesn't have a use anymore

brunofalmeida, 2018-07-26:
All of the things left to be tested and merged from pre-qual work were made into issues.

siddMahen, 2018-07-26:
Ok, good to know. @jackberezny I'm looking at the UART test code on `qual_assert` and it looks like you're the only one whose every touched the file via git. Were you working on it pre-qual?

jackberezny, 2018-07-26:
@siddMahen I didn't write any of the UART code; the file was taken directly from @Tahreeem.

siddMahen, 2018-07-27:
I've currently assigned the UART testing to @Tahreeem, so I just wanted to make sure that was right. Thanks Jack, Bruno.

closed 2018-10-12 by brunofalmeida