utat-ss/HERON-lib-common #59: ADC + conversion interface

siddMahen, 2018-07-25:
@brunofalmeida Looking at the ADC + coversion code. What do you think of the following interface?

Two key structs: a central `adc_t shared_adc` which manages the connection to the ADC via SPI, etc, and a bunch of `src_t` objects, where

```
typedef struct {
    adc_t* adc;
    uint8_t channel;
    uint16_t raw_data;
    uint16_t data;
    transform_fn_t transform_fn;
} src_t;
```

and `transform_fn_t` is a function that takes 12-bit raw ADC values as input, and returns a value with "real" units.

Here's how I'm imagining this could be used:

```
#include <adc/adc.h>
#include <adc/eps.h>

transform_fn_t* raw_to_voltage; // conversion fn declared in adc/eps.h and defined in libadc
transform_fn_t* raw_to_current; // ^^ same 

adc_t shared_adc;
src_t eps_voltage;
src_t eps_current;

int main() {
    init_adc(&shared_adc, ADC_EPS);
    init_src(eps_voltage, raw_to_voltage, EPS_VOLTAGE_CHANNEL);
    init_src(eps_current, raw_to_current, EPS_CURRENT_CHANNEL);

    // ...

   double v = read(&eps_voltage);
   double i  = read(&eps_current);
   double pwr = i*v;

   // ...
}

```

siddMahen, 2018-07-25:
@jadenreimer @timmysilv @hirolemo Would something like this suite your needs? If not, what's lacking?

jadenreimer, 2018-07-26:
I can't speak for the others but I currently don't know enough about how we will use the ADC software-wise to give a satisfactory answer. Let me see if I can make more sense of it looking again at our current ADC code.

brunofalmeida, 2018-07-26:
@siddMahen How would this work for doing data conversions in repositories that don't actually use an ADC device?

siddMahen, 2018-07-26:
@brunofalmeida I'm imagining that when data is collected on 32m1 A, and needs to be sent/processed to/in 32m1 B, it's always converted from a raw measurement to a final value before being sent to B.

siddMahen, 2018-07-26:
Oh, sorry, I think I misunderstood; are you asking how this system would handle data/data sources whose raw source isn't an ADC? Because of the channel requirement, I don't think this exact setup would be applicable in those situations, but I think that it could be easily tailored to a particular underlying source.

For example, maybe we'd want to wrap all raw sources in this way, something like:

```
    x_raw_data_t raw_source_x;
    x_src_t src_assoc_to_x;
    // ...
    init_raw_x(&raw_source_x);
    init_src_x(&src_assoc_to_x, conversion func, additional data maybe);
    // ...
    double val = read(&src_assoc_to_x);
```

vogeldylan, 2018-07-26:
This last point, about wrapping all raw sources in some kind of wrapper with a corresponding conversion function is interesting. But here, let me mention some use cases on the sat:

* There are a variety of raw data sources in the satellite. Some, like the EPS ADC, and the PAY-SSM ADC actually have different conversion functions for specific channels. i.e. some PAY-SSM ADC channels measure temperature, some measure light intensity, etc. EPS does voltage, current and maybe some other value.
* _Some_ of these raw values need to be converted on a micro to perform decision making. i.e. EPS may make some decisions based on current draw.
* Some of these raw values do not need to be converted. i.e. PAY-SSM makes no decisions based on environmental data or optical readings. We just return values.
* Based on the above, the system you imagine _could_ be expanded to include a pointer to some 'get data' function, and another pointer to some 'convert data' function. i.e. OBC wants the temperature value from PAY to make some decisions. OBC's struct could actually call a function to fetch the raw data over CAN, and then pipe it through the conversion function it defines. 
* The last point would also allow this kind of interface to be expanded to pressure sensors, humidity, other generic input raw data and output some converted value.

Not sure if this helps. Ramblings of an old man who hasn't used the recently developed ADC functions.

vogeldylan, 2018-07-26:
Also, sending raw values around should save some processor cycles and maybe size in memory. Some of those conversions give us huge double values from 14-24 bit input.

siddMahen, 2018-07-26:
@vogeldylan I think this is an idea worth pursuing. Here is an example of the kind of interface we could provide:

```
// src_t are generic sources (adc, thermistor, env sensor, etc), which return the raw
// data from a data source.
typedef struct {
    // public fields
    fetch_data_fn_t fetch;

    // private fields
    uint8_t pin;
    port_t ddr;
    port_t port;
    board_t board;
    // some additional device specific info
} src_t;

// public methods
uint32_t read_src(src_t* src);

// sink_t's are generic data sinks; these draw data from src_t's and consume them in some way (either // raw, or through some conversion/transformation
typedef struct {
    src_t* src;
    transform_fn_t transform; // the transform function has signature uint32_t, additional data -> double 
    // additional info perhaps
} sink_t;

// public methods
double read(sink_t* sink);
uint32_t read_raw(sink_t* sink);
```

brunofalmeida, 2018-07-26:
What I meant was that as far as I'm aware, all measurements have raw data (a set of bits represented as a `uint`) that must be converted to a double value. Unless someone can figure out otherwise, we don't have a way of sending double values over CAN so we have to send the raw data and do the conversion on the other side that wants to use it.

brunofalmeida, 2018-07-26:
So for example, OBC would receive the raw data from PAY's ADC and have to convert that itself without actually having the ADC device that the data comes from.

brunofalmeida, 2018-07-26:
Also, how would this source concept work for data sources that have different sizes of data?

siddMahen, 2018-07-26:
I think part of Dylan's idea was to allow, for example, the `src_t`s to be CAN sources on some other board. So the `src_t`s would do all the CAN legwork required to get a new value.

brunofalmeida, 2018-07-26:
How would that work since CAN is asynchronous?

brunofalmeida, 2018-07-26:
How could we avoid a bunch of code duplication for cases such as having lots of ADC channels with identical conversions?

siddMahen, 2018-07-27:
> how would this unified source concept work for various sources that have different sizes of data?

I think we can just create a union struct with the largest possible size, probably uint64_t, and just use that.

> How would that work since CAN is asynchronous?

For the CAN case (or maybe more generally the async case) we'd just block until the value arrived; after all, that's what'd need to happen anyways if you needed to access data from another board.

> How could we avoid a bunch of code duplication ... [e.g.] channels with identical conversions?

I think if we made judicious use of `enums`, function pointers, and good defaults, this could be fairly simple. We'd need to tack on some additional data, like channel number for example, but we could do this in an adhoc way without too much trouble.

siddMahen, 2018-07-27:
I think this would be easier to evaluate with more concrete examples and more concrete code. Once we have the adc working, and at least one pay source working, building this becomes more practical.

closed 2018-09-15 by brunofalmeida