utat-ss/HERON-lib-common #61: Implement a more efficient channel reading algorithm in ADC lib

siddMahen, 2018-07-26:
> Implement more efficient channel reading algorithm in lib-common ADC library, instead of having to waste SPI frames from waiting for next message.

Quoted verbatim from the pay/thermal branch thermistor code base.

vogeldylan, 2018-07-26:
[image]

vogeldylan, 2018-07-26:
Took a look at the code, saw what you were talking about. SPI messages are sent until the correct channel address is pre-pended to the outgoing message. It's a good safety check, but according to the data sheet this should be the very next frame.

Can someone test this? Every time a spi_transfer is wasted waiting for the conversion, output some message to UART, or increment a counter? My suspicion is that it'll only take one frame to return.

vogeldylan, 2018-07-26:
If we're polling large amounts of data, we could also lower the number of SPI frames from 2 -> 1 if we send a new channel address every frame, and then sort the returned data by the corresponding address.

siddMahen, 2018-07-26:
@vogeldylan I think if we keep track of the last 2 frames of data, we can significantly increase efficiency. From what I understand, the datasheet says that when you request a reading from channel x (on frame n), you get the corresponding reading 2 frames *later*, on frame n + 2. Meanwhile, the n-th frame is returning data from the (n - 2)-th request, and the (n + 1)-th frame is returning data for the (n - 1)-th request etc.

siddMahen, 2018-07-26:
Another idea: use the auto-1 (or auto-2) mode to just cycle through all the channels at some fixed interval, and collect the readings in an array. When a reading for channel i is request, and if its not too stale, return the cached version. Otherwise, refresh all the readings and return the new value.

siddMahen, 2018-07-26:
The above idea could also be used in tandem with turning the ADC on and off. Turn it on when we need it, otherwise, keep it off.

siddMahen, 2018-07-26:
If we used an auto-1 mode, I could imagine the following very slick interface:
```
adc_t shared_adc = {
    .channel = 0b01000101 // initiate auto 1 mode with channels 0, 2 and 6
    // pin data for SPI, CS etc etc
}

init_adc(&shared_adc);

uint16_t raw_2 = read_channel(&shared_adc, 2); // poll all channels 0, 2 and 6, then shutdown

//.. later

uint16_t raw_6 = read_channel(&shared_adc, 6); // no need to poll! return cached value of chan 6
```

vogeldylan, 2018-07-27:
Seems like a reasonable interface, especially for batch reads (i.e. we might only be returning sensor now over CAN, but we know the next CAN request will be for the next channel)

> From what I understand, the datasheet says that when you request a reading from channel x (on frame n), you get the corresponding reading 2 frames later, on frame n + 2.

This is incorrect. Check the n+1 box on the flow-chart. SDO returns conversion data of channel selected in frame n.

siddMahen, 2018-07-27:
@vogeldylan Yes you're right; I was reading about manual mode; in auto mode, you get the conversion data on the next frame.

siddMahen, 2018-07-27:
Uploaded a preliminary version of the auto-1 based adc implementation on the `adc-interface` branch @vogeldylan @brunofalmeida check it out. Also rewrote `func_tests/adc_test.c` to be compatible with the new interface:

```C
#include <uart/uart.h>
#include <adc/adc.h>

#ifndef F_CPU
#define F_CPU 8000000L
#endif

#include <util/delay.h>

adc_t adc = {
    .channels = 0x0fff; // poll all 12 channels
    .port = ADC_CS_PORT_EPS;
    .ddr = ADC_CS_DDR_EPS;
    .pin = ADC_CS_PIN_EPS;
};

// This test reads the raw data and voltages on each ADC channel
int main(void) {
    init_uart();
    print("UART initialized\n");

    init_adc(&adc);
    print("ADC initialized\n");

    while (1) {
        fetch(&adc); // fetches all 12 channels worth of data

        for (uint8_t i = 0; i < ADC_CHANNELS; i++) {
            uint16_t raw_data = read_channel(&adc, i);
            double raw_voltage = adc_convert_raw_data_to_raw_voltage(raw_data);
            print("Channel: %u, Raw Data: 0x%.4x, Raw Voltage: %lf\n",
                    i, raw_data, raw_voltage);
        }

        _delay_ms(200);
    }
}
```

siddMahen, 2018-07-27:
The library compiles but that example probably wont compile; need to add the right header for the `adc_convert_raw_data_to_raw_voltage` f'n.