utat-ss/HERON-lib-common #84: Prevent EEPROM corruption

brunofalmeida, 2018-11-15:
Refer to section 8.4.2 of the 32M1/64M1 datasheet. We probably need to enable brownout detection.

jackberezny, 2019-06-29:
From section 8.4.2 of the data sheet:
> Keep the AVR RESET active (low) during periods of insufficient power supply voltage. This can be done by enabling the internal Brown-out Detector (BOD). If the detection level of the internal BOD does not match the needed detection level, an external low VCC reset Protection circuit can be used. If a reset occurs while a write operation is in progress, the write operation will be completed provided that the power supply voltage is sufficient.

Per section 28.2 on fuse bits, BODLEVEL0, BODLEVEL1 and BODLEVEL2 bits are set as 1 by default and can be used to change the brown-out detection level based on the table on pg. 399 (section 29.6). Based on some preliminary searches, it looks like an acceptable range would be between 4.3-4.5V.

However, we still have the potential issue of the brownout-detection being triggered during a write, at which point the data could very likely be corrupted. I am thinking that we could either implement some degree of data redundancy or add a checksum onto existing data (e.g. CRC).

@brunofalmeida

brunofalmeida, 2019-12-22:
We'll set the BODLEVEL fuses so brownout detection is enabled with a threshold of 2.8V. This is now specified in the Software Deployment Procedure document.

closed 2019-12-22 by brunofalmeida